asiermarzo/Ultraino #1: Wrong scaling of the spherical cap .stl

kevinscellnano, 2018-01-08:
When the file from:
https://github.com/asiermarzo/Ultraino/blob/master/arrays/spherical%20cap.STL

is imported into FreeCAD or Ultimaker Cura the model is 25 times smaller than it should be, 
for the 1cm transducers to fit into their places.

I was wondering, if this is due to an import error or if the .stl has a false scaling.

Best regards, 
Kevin

asiermarzo, 2018-01-08:
Hi kevin,
I apologize some the STLs still use strange scaling due to inch/mm.
The factor is 25.4 I have updated the STL file.
Thanks for the notice.
Best,
Asier.

kevinscellnano, 2018-01-08:
Hy Asier,

Thank you for the quick response. The updated .stl is now scaled correctly in Cura.

Best regards, Kevin

asiermarzo, 2019-03-29:
Things seems to work now

closed 2019-03-29 by asiermarzo