asiermarzo/Ultraino #2: Spherical Cap.xml.gz does not fit to Spherical Cap.STL

PhilippHoe, 2018-05-18:
Hi,
i tried building the Spherical Cap Array but the xml.gz file does not fit to the STL file because in die STL file you use 64 transducers an in the xml.gz file you only use 45 transducers.
Can you please generate the fitting xml.gz array with 64 transducers for me?

Greetings from germany

Philipp

asiermarzo, 2018-05-18:
I think now there are simulations and STL for both spherical Caps. Best.

PhilippHoe, 2018-05-18:
Thank your very much!
But i think there ist still a difference between the simulation file and the STL file... In the STL file the second ring of the array has only 7 transducers and in the simulation file the second ring has 8 transducers.

Greetings
Philipp

asiermarzo, 2018-05-20:
I think now the simulation file will fit the STL. best.

closed 2019-03-29 by asiermarzo