asiermarzo/Ultraino #3: DriverNano16

vrossbach, 2018-09-27:
I am trying to upload the code for DriverNano16 in the Arduino Uno and this error messages appear:

'class HardwareSerial' has no member named '_dataAvailable'
'class HardwareSerial' has no member named '_peekData'

I tried to put the functions Serial.available() and Serial.peek() but then there is a error to compile to Arduino Uno board. Is there a different code for Arduino Uno or the code was written in another version?

asiermarzo, 2018-09-27:
Try the instructions on https://github.com/asiermarzo/Ultraino/blob/master/Arduino%20MEGA%2064%20driver%20board%20code/readme.txt

Best

vrossbach, 2018-09-27:
Hi, 

now I got, I had not pasted the code in the correct public position.

Thank you!

closed 2019-03-29 by asiermarzo