asiermarzo/Ultraino #22: Noise on signal when doing semi-manual pin assignment, software frozen

littlejacinthe, 2020-04-03:
Hello, 

When doing the semi-manual pin assignment following the steps of your video, we noticed that :
- The Ultraino software stops working at : 
Transducer Assignment menu > Connect (Arduino Nano is connected correctly) > Start > Check.  Once we click on the 'check' command there is no output when it should show numbers for Selected, Pin, Amplitude, Phase. After that the software is completely frozen and we have to restart it. 
It probably is because the software does not detect the signal coming from the transducers correctly, as when we checked it with the oscilloscope : 
- There is noise provided by the transducer used to check the signal when it's connected to wires. We tried different wires but there was always noise, which doesn't happen when the transducer is directly connected to the probe of the oscilloscope.

We used all materials recommended as well the codes provided during this experiment. 
Our driver is an Arduino MEGA, and we are working on an 8x8 flat array. 
You can see how we proceeded on [this video](https://youtu.be/fm1WUwtMGUc). It shows more precisely the problem that we're having. 

Would you have any idea why is this happening and how to solve the problem ?

Thank you

asiermarzo, 2020-04-08:
Hi,
Very good work with the video and with the hardware. Ultraino started as research and still needs a lot of brushing up.

a) Have you installed the phase-detector sketch in the arduino that is used to detect the pins?
b) when you connect the driverboard and the phase-detector have you identified which COM port is which?
c) be sure to connect the main program to the board and the connect of the assignments to the phase detector.

In any case, I will try to add more debug messages and avoid the software from hanging.

Best,
Asier.

littlejacinthe, 2020-04-10:
Hello, 

a) we installed the phase-detector sketch using the same materials 
[image]
b) and c) We made the mistake of connecting the wrong ports before but corrected it and then this noise problem came up. 

We are going to order brand new materials for the phase detector as the ones we've been using are not and there might be a default there. 

Thanks for your reply !

littlejacinthe, 2020-04-16:
Hello, 

After changing the hardware of the arduino phase detector, we still got the same error. 

I noticed when connecting the Arduino Nano that we have to use the old bootloader for ATMega 328P otherwise it doesn't upload the code. It seems like the bootloader was updated in 2018. Could it be that you used the new bootloader ? 

Also on line 34 of the phase detector code, the line should print the samples array. When the board is connected to the simulation software it says the port is busy, and when the board is not connected to the simulation software (with only the Mega board sending the signal to the transducers) the monitor is empty. Is that normal ?

Thanks

asiermarzo, 2020-09-13:
It is not possible to have the phase detector working and the serial monitor connected to it. Have you selected the transducers in the order that you want them tested and then pressed the button Order? Otherwise, all of them may have order 0 and cause problems. Best. I know the phase assignment works like shit, I need to brush it up asap.